SashaBrooke/OhPossum #46: M30 thread test print

SashaBrooke, 2025-03-30:
**Hardware request description:**
@joeyboy2023 to model and print samples of internal and external M30 thread. This should be done with and without supports to determine the best way to print the axle shaft and bottom plate for the single axis test jig.

SashaBrooke, 2025-04-04:
Apparently shouldn't require supports

SashaBrooke, 2025-04-04:
Test prints should include face offsets and filleting of thread peaks and valleys. May be worth using the test prints to check fitment and determining the best combination of these factors to ensure the Axle Shaft and Bottom Plate mate correctly in the Single Axis Test Jig.

SashaBrooke, 2025-04-15:
Test print with no face offset complete - far too tight.

joeyboy2023, 2025-04-19:
Test prints completed:
0mm offset = too tight
0.1mm offset = too tight
0.1mm offset + fillet = too tight with printing error, once corrected was a nice fit with little resistance and play
0.15mm offset + fillet = no resistance but a bit too much play

closed 2025-04-19 by joeyboy2023