SashaBrooke/OhPossum #47: Single axis test jig top plate redesign - PCB mounting

SashaBrooke, 2025-03-30:
**Hardware request description:**
Top plate needs to have mounting standoffs added to provide a way of mounting the control system microcontroller and motor driver board PCBs. Theoretically are not necessary (could just drill holes) but good practice and a similar design will be needed in the final prototype anyway.

joeyboy2023, 2025-04-05:
PCB mount holes were all similar as measured to imported CAD file.
Raspberry Pi was similar longwise, the holes on the short side had approx. 0.2mm variation when comparing imported CAD file to physical board. 

**Action:** Slots will need to be made for the holes on the short side

joeyboy2023, 2025-04-19:
Also needs mounting holes for the Raspberry Pi debug probe board

SashaBrooke, 2025-04-25:
Debug probe will be added in a future rework.

SashaBrooke, 2025-04-25:
Need to investigate possible mounting issue - motor driver board standoffs don't seem to line up with holes on the PCB.

*Left pair of standoffs (below)*

<kbd>[image]</kbd>

SashaBrooke, 2025-04-25:
Also add fasteners once above concern in resolved

closed 2025-04-27 by SashaBrooke